makerbase-mks/MKS-TinyBee #111: Layer shift when refreshing web-page

TjeerdvanDijk, 2024-02-06:
I enabled esp3d wifi support but every single time i refresh the page i get a layer shift.

TjeerdvanDijk, 2024-02-06:
here is my config
[configuration.h](https://hastebin.com/share/ejedohenek.cpp)
[configuration_h](https://hastebin.com/share/comibexeqa.cpp)

PanosPetrou, 2024-03-01:
Interacting with the web interface during printing, forces the processor to allocate some unplanned resources there and messes up printing. So don't use the web interface during that time.

josh2595, 2025-05-05:
I experience layer shifting during printing on the x and y axes on my custom-made 3d printer using the MKS Tinybee board. At first, I thought it was the TMC2209 driver vref wrong adjustment, which led to skipped steps, so I tried multiple vref adjustments from 0.72 - 1.2, but still the same problem exists. I don't have no idea what causes it

PanosPetrou, 2025-05-06:
1. Don't use the web interface during printing .
2. On Configuration.h, on X_DRIVER_TYPE, Y_DRIVER_TYPE , Z_DRIVER_TYPE if you set them to TMC2209_STANDALONE and still layer shits are happening, try to set them to A4988 or DRV8825.
3. On Configuration.h, keep #define PRINTCOUNTER disabled / commented out.

josh2595, 2025-05-06:
I have done all this, but still I experience layer shift

[image]
[image]

josh2595, 2025-05-10:
I realized that the layer shifts in the X and Y axes were caused by loosely tightened threaded rods on the Z axis. When printing at high speeds, these rods tended to shake. After tightening the grub nut, the problem was resolved.

The only issue I am experiencing now is that sometimes at the beginning of the print, the extruder motor tends to rotate in the opposite direction. This causes the filament to unload instead of being pulled in for printing. This problem occurs occasionally, but not every time I print.