makerbase-mks/MKS-TinyBee #117: Motors not working

imbence, 2024-04-18:
Hi,
Just got the MKS Tinybee with an MKS TFT35 screen and A4988 drivers. I connected them up with 3 motors for testing but I'm unable to make the motors move with the default firmware as it came with.
Can someone help me with what could be the problem?

[image]

PanosPetrou, 2024-04-22:
I have the same board, but not the same screen, so I can't help you fully. I would connect via USB and use Pronterface / Printrun to move the motors using gcode commands.

Some things to keep in mind:
A lot of stuff on your board are not connected and that might generate errors preventing motor movement.
That screen is often given as a set with a different board, I'm not sure how compatible it is with Tinybee.
As far as I know, Tinybee is shipped (on new orders) empty, without Marlin installed. Did you configure & upload it correctly?

KillerInk, 2025-01-08:
you have plugged your steppers wrong.

joaomarcosvilar, 2025-05-11:
Hello! How are you? I'm having problems with my code to communicate between the MKS tinybee and the TFT35, could you guide me to program it correctly?

gregio1, 2026-03-23:
@joaomarcosvilar 

Hi! I'm having the **same TFT35 communication problem** with MKS TinyBee.  

Did you solve it and **how**?  

**My setup:**
- TinyBee V1.0 + MKS TFT35 V1.0 (FW 1.0.8)  
- UART connected: TX→GPIO17(RX2), RX→GPIO16(TX2), GND  
- USB works, TFT buttons = dead