makerbase-mks/MKS-TinyBee #125: followed instructions and fails to compile with marlin 2.1.2.5 bugfix

theyeetmaster992, 2025-01-26:
Unable to be complied and uploaded. Always end with this error code in terminal.

Marlin\src\libs\numtostr.cpp:28:32: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
 #pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
                                ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
Marlin\src\inc\Warnings.cpp:74:4: warning: #warning "Your Configuration provides no method to acquire user feedback!" [-Wcpp]
   #warning "Your Configuration provides no method to acquire user feedback!"
    ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncPrinter.h:51:0,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncPrinter.cpp:51:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:104:0:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W_buffer.h:60:0,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W_buffer.cpp:58:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.cpp:53:0:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/opt.h:51:0,
                 from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwip/ip_addr.h:40,
                 from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\tcpip_adapter/tcpip_adapter.h:45,
                 from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/esp_event_legacy.h:23,
                 from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_event/esp_event.h:26,
                 from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32/esp_event_loop.h:22,
                 from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiGeneric.h:27,
                 from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFiSTA.h:28,
                 from C:\Users\Server\.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src/WiFi.h:32,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:84,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.cpp:53:
C:\Users\Server\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\lwip/lwipopts.h:326:0: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_TCP_MSS
 ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.h:60:0,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.cpp:52:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src/async_config.h:35:0: note: this is the location of the previous definition
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
 ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mks_tinybee\lib1b4\AsyncTCP_RP2040W\AsyncPrinter.cpp.o] Error 1
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:331:8: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   addr.addr = ip;
        ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)':
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:390:42: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
     returnValue = connect(IPAddress(addr.addr), port);
                                          ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:109:0,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:104:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:393:73: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
     ATCP_LOGDEBUG3("connect: dns_gethostbyname => IP =", IPAddress(addr.addr), ", returnValue = ", returnValue ? "TRUE" : "FALSE");
                                                                         ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W_Debug.h:87:128: note: in definition of macro 'ATCP_LOGDEBUG3'
 #define ATCP_LOGDEBUG3(x,y,z,w)  if(_ASYNCTCP_RP2040W_LOGLEVEL_>3) { ATCP_PRINT_MARK; ATCP_PRINT(x); ATCP_PRINT_SP; ATCP_PRINT(y); ATCP_PRINT_SP; ATCP_PRINT(z); ATCP_PRINT_SP; ATCP_PRINTLN(w); }
                                                                                                                                ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::operator==(const AsyncClient&)':
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:468:66: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return (_pcb != NULL && other._pcb != NULL && (_pcb->remote_ip.addr == other._pcb->remote_ip.addr) &&
                                                                  ^
compilation terminated due to -fmax-errors=5.
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src/async_config.h:35:25: error: 'MBED_CONF_LWIP_TCP_MSS' was not declared in this scope
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.h:86:34: note: in expansion of macro 'TCP_MSS'
     SyncClient(size_t txBufLen = TCP_MSS);
                                  ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src/async_config.h:35:25: error: 'MBED_CONF_LWIP_TCP_MSS' was not declared in this scope
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.h:87:55: note: in expansion of macro 'TCP_MSS'
     SyncClient(AsyncClient *client, size_t txBufLen = TCP_MSS);
                                                       ^
*** [.pio\build\mks_tinybee\lib1b4\AsyncTCP_RP2040W\AsyncTCP_RP2040W.cpp.o] Error 1
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mks_tinybee\lib1b4\AsyncTCP_RP2040W\AsyncTCP_RP2040W_buffer.cpp.o] Error 1
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mks_tinybee\lib1b4\AsyncTCP_RP2040W\SyncClient.cpp.o] Error 1

WillyECoyote77, 2025-01-30:
Hello!

I wanted to give my old companion Anet A8 a "brain transplant" with MKS TinyBee V1.0 board and I hit the wall with exactly the same problem. I had to settle with 2.0.9.2 version for now.

I'm willing to sacrifice WiFi connectivity, if turning it off in the code would mean, that I can compile 2.1.2.5 bugfix version. Sadly I do not have a clue what and where in the code can/should be commented out to temporary bypass the issue, if at all possible.

Regards

theyeetmaster992, 2025-01-30:
> Hello!
>
> I wanted to give my old companion Anet A8 a "brain transplant" with MKS TinyBee V1.0 board and I hit the wall with exactly the same problem. I had to settle with 2.0.9.2 version for now. [...]

So far i have found one more up to date. Still get warnings. Wifi you uncomment in configuration_adv.h, you can use the "find" option in vscode to help you find it. I have also started the try older version of the builder to try and get it working. Seems slightly better but still get warnings. Compiling is possible but not sure about flash.

Not sure what the right fix is since i don't really know anything about code like this. Machine i need it for is currently still standing and with no progress. There must be a way!

https://github.com/luxwarp/Marlin/tree/b35e8f28435d8c7364100a5fa01e54cb0ef71267/Marlin
^chad guy right here

theyeetmaster992, 2025-01-30:
Still get this then creates a "partitions.bin" file

In file included from C:/Users/natopc/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:146:0,
                 from Marlin/src/core/../inc/../HAL/./ESP32/../shared/Marduino.h:36,
                 from Marlin/src/core/../inc/../HAL/./ESP32/HAL.h:30,
                 from Marlin/src/core/../inc/../HAL/HAL.h:30,
                 from Marlin/src/core/../inc/MarlinConfig.h:31,
                 from Marlin/src/core/../MarlinCore.h:24,
                 from Marlin/src/core/utility.cpp:25:
C:/Users/natopc/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/WString.h:39:0: warning: "FPSTR" redefined
 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
 ^
In file included from Marlin/src/core/utility.h:25:0,
                 from Marlin/src/core/utility.cpp:23:
Marlin/src/core/../core/types.h:32:0: note: this is the location of the previous definition
   #define FPSTR(S) (reinterpret_cast<FSTR_P>(S))
 ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncPrinter.h:51:0,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncPrinter.cpp:51:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:104:0:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W_buffer.h:60:0,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W_buffer.cpp:58:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.cpp:53:0:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\tools\sdk\include\lwip/lwip/opt.h:51:0,
                 from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\tools\sdk\include\lwip/lwip/ip_addr.h:40,
                 from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\tools\sdk\include\tcpip_adapter/tcpip_adapter.h:45,
                 from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\tools\sdk\include\esp32/esp_event_legacy.h:23,
                 from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\tools\sdk\include\esp_event/esp_event.h:26,
                 from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\tools\sdk\include\esp32/esp_event_loop.h:22,
                 from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\libraries\WiFi\src/WiFiGeneric.h:27,
                 from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\libraries\WiFi\src/WiFiSTA.h:28,
                 from C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\libraries\WiFi\src/WiFi.h:32,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:84,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.cpp:53:
C:\Users\natopc\.platformio\packages\framework-arduinoespressif32@3.10004.210126\tools\sdk\include\lwip/lwipopts.h:326:0: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_TCP_MSS
 ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.h:60:0,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.cpp:52:
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src/async_config.h:35:0: note: this is the location of the previous definition
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
 ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:331:8: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   addr.addr = ip;
        ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)':
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:390:42: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
     returnValue = connect(IPAddress(addr.addr), port);
                                          ^
In file included from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.h:109:0,
                 from .pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\AsyncTCP_RP2040W.cpp:104:
[... cut ...]
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src/async_config.h:35:25: error: 'MBED_CONF_LWIP_TCP_MSS' was not declared in this scope
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^
.pio\libdeps\mks_tinybee\AsyncTCP_RP2040W\src\SyncClient.h:87:55: note: in expansion of macro 'TCP_MSS'
     SyncClient(AsyncClient *client, size_t txBufLen = TCP_MSS);
                                                       ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mks_tinybee\libfc8\AsyncTCP_RP2040W\SyncClient.cpp.o] Error 1

theyeetmaster992, 2025-01-30:
What i get when i use marlin fork i posted earlier.

o run --silent -e mks_tinybee
Marlin/src/feature/powerloss.cpp: In static member function 'static void PrintJobRecovery::resume()':
Marlin/src/feature/powerloss.cpp:606:75: warning: format '%ld' expects argument of type 'long int', but argument 3 has type 'uint32_t {aka unsigned int}' [-Wformat=]
   sprintf_P(cmd, PSTR("M24S%ldT%ld"), resume_sdpos, info.print_job_elapsed);
                                                                           ^
Marlin/src/feature/powerloss.cpp:606:75: warning: format '%ld' expects argument of type 'long int', but argument 4 has type 'millis_t {aka unsigned int}' [-Wformat=]
Marlin/src/feature/powerloss.cpp:606:75: warning: format '%ld' expects argument of type 'long int', but argument 3 has type 'uint32_t {aka unsigned int}' [-Wformat=]
Marlin/src/feature/powerloss.cpp:606:75: warning: format '%ld' expects argument of type 'long int', but argument 4 has type 'millis_t {aka unsigned int}' [-Wformat=]
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'

theyeetmaster992, 2025-01-30:
This seems to be a good bet for now. https://github.com/LPR-Michael/MKS-TinyBee-for-Artillery-Hornet

theyeetmaster992, 2025-01-31:
[Marlin - 2.zip](https://github.com/user-attachments/files/18620132/Marlin.-.2.zip)

Place main marlin bugfix 2.1.x folder in C:\ for best effect. Here is my cfg files. neo pixel is also set up. Just replace my machines values with your machines values.

WillyECoyote77, 2025-02-03:
Tnx! Sorry for my late reply. I'll try it out and report.

vcijj, 2025-02-12:
我遇到了同样的问题，我才用的方式是删除新增的CYW43439文件，这明显是marlin官方的一个错误，删除此文件后编译上传成功。

WillyECoyote77, 2025-02-15:
> 我遇到了同样的问题，我才用的方式是删除新增的CYW43439文件，这明显是marlin官方的一个错误，删除此文件后编译上传成功。

I can't find this file.

theyeetmaster992, 2025-02-16:
> "_in C:\ for best effect_" - thanks for the tip theyeetmaster992. Root folder sure does help.
> 
> What I found out trying to upload **Marlin 2.1.2.5 with bug fixes** that the moment I enable serial port 2 compiling error occurs. _f:\marlin-2.1.x\marlin\src\core\serial.h:137:55: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT {aka int}'_
> 
> For what is serial port 2 used? Wifi?
> 
> Regards

Yes Serial port 2 is used for Wifi

WillyECoyote77, 2025-02-16:
Did you manage to compile and upload with working WiFi configuration? I some other post you were mentioning ESP32 GitHub resource that helped you out. Can you please provide link?

Regards

mojtabahdpr, 2025-03-01:
hi 
i have this probhlem 
do you solve it ?

WillyECoyote77, 2025-03-01:
Sadly no. After investing quite some time trying to solve this, I gave up for now.

I wish that someone who managed to make it work, would write down a step by step instructions in concise and understandable manner. Posting modified config files is a nice gesture, but if you do not know what and why is changed, you are doomed to check and try out all different configurations scattered in different posts until you find the working one. That takes a lot of time and frustration.

theyeetmaster992, 2025-03-01:
Got board back from electronic repair friend. He says its ok, will do a test with it. Only endstops where not working(uses optical) Sent from my Huawei Mobile

Pattis74, 2025-06-25:
I just managed to compile the version marlin 2.1.2.5 for the MKS TinyBee.
As the error messages said, multiple libraries are found for "AsyncTCP" and "ESPAsyncTCP".
I downloaded the older version for the MKS TinyBee. marlin 2.0.9.2.
It can be compiled with some warnings.
Then I copied al the working libraris from the ".pio\libdeps\mks_tinybee\" folder to the marlin 2.1.2.5 ".pio\libdeps\mks_tinybee\" folder.
Then I could compile it with the same warnings, like in the older version.
I uploaded it to the board.
After uploading the WebUI and setting the WIFI configuration everything worked fine.
I think, some ini files need to be adjusted to grab the right libraries in marlin 2.1.2.5.

theyeetmaster992, 2025-07-05:
> Did you manage to compile and upload with working WiFi configuration? In some other post you were mentioning ESP32 GitHub resource that helped you out. Can you please provide link?
> 
> Regards

https://github.com/luc-github/ESP3DLib

0x5e, 2025-07-09:
I found `mks tinybee marlin/ini/features.ini` didn't specific version for `WIFISUPPORT` libs.

Change `mks tinybee marlin/ini/features.ini` WIFISUPPORT section to:
```
(ESP3D_)?WIFISUPPORT                   = me-no-dev/AsyncTCP@1.1.1, me-no-dev/ESP Async WebServer@1.2.3
                                         ESP3DLib=https://github.com/luc-github/ESP3DLib/archive/master.zip
                                         arduinoWebSockets=links2004/WebSockets@2.3.4
                                         luc-github/ESP32SSDP@^1.1.1
                                         lib_ignore=ESPAsyncTCP
```

The version above matches `.pio` folder from git. 
Then remove `.pio` and build twice (first time failed), then succeed for me.

Pattis74, 2025-07-09:
Sadly this still did not worked for me.

1.: the lib me-no-dev/ESP Async WebServer@1.2.3 can not be downloaded:

`Library Manager: Installing me-no-dev/ESP Async WebServer @ 1.2.3
Warning! Could not find the package with 'me-no-dev/ESP Async WebServer @ 1.2.3' requirements for your system 'windows_amd64'
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Adafruit NeoPixel @ 1.8.7
|-- U8glib-HAL @ 0.5.4
|-- AsyncTCP @ 1.1.1
|-- ESP3DLib @ 1.0.12
|-- WebSockets @ 2.3.4
|-- ESP32SSDP @ 1.1.1
|-- SPI @ 1.0
|-- EEPROM @ 1.0.3
|-- ArduinoOTA @ 1.0
|-- ESPmDNS @ 1.0
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- SPIFFS @ 1.0
|-- Wire @ 1.0.1`

I tested it also on Linux with no luck.

The current library contains also code for the rp2040w, wich I can not compile.

But it can still be downloaded from here:
[https://github.com/makerbase-mks/MKS-TinyBee/tree/main/firmware/mks%20tinybee%20marlin/.pio/libdeps/mks_tinybee](url)

2.: there is an error in the path to the HAL libraries:
`.pio/libdeps/mks_tinybee/ESP3DLib/src/espcom.cpp:25:52: fatal error: ../../../../../Marlin/src/HAL/HAL/ESP32/FlushableHardwareSerial.h: No such file or directory
compilation terminated.
*** [.pio\build\mks_tinybee\lib225\ESP3DLib\espcom.cpp.o] Error 1`

The file needs to be changed: ".pio\libdeps\mks_tinybee\ESP3DLib\src\esp3dlibconfig.h"

`#ifndef HALHOME`
`#define HALHOME SRCHOME`
`#endif`

Then I was able to compile it again.

nguyenchiencuong1801-sketch, 2025-09-05:
tôi đã gỡ cài đặt thư viện Async  rp2040w và biên dịch thành công