makerbase-mks/MKS-TinyBee #134: Btt mini 12864 config tinybee

akhteh, 2025-04-07:
My friends, no matter what I do, the screen does not turn on, the firmware is done correctly and the axes are controlled from the computer Only the display does not turn on with tinybee Display btt mini 12864 v2 Please help if you know
The cables have been fixed The screen is healthy

[image]

PanosPetrou, 2025-05-06:
Perhaps the pin-out of your screen is different, have you set the correct model on Marlin? I'm using MKS Mini 12864 V3 without issues.

akhteh, 2025-05-06:
> شاید پین‌های خروجی صفحه نمایش شما متفاوت باشد، آیا مدل صحیح را روی Marlin تنظیم کرده‌اید؟ من از MKS Mini 12864 V3 بدون مشکل استفاده می‌کنم.

I even bought MKS Mini 12864 V3, still with this display, z offset cannot be given during printing, please send me your firmware as well.

akhteh, 2025-05-06:
> Perhaps the pin-out of your screen is different, have you set the correct model on Marlin? I'm using MKS Mini 12864 V3 without issues.

I even bought MKS Mini 12864 V3, still with this display, z offset cannot be given during printing, please send me your firmware as well.