makerbase-mks/MKS-TinyBee #138: Problem with compilation

torys877, 2025-06-13:
I used t;his repo about 6-8 month ago and all was good, but now I cannot compile project, because of error:

I run:
`pio run`

```
In file included from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncPrinter.h:51:0,
                 from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncPrinter.cpp:51:
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:104:0:
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W_buffer.h:60:0,
                 from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W_buffer.cpp:58:
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/SyncClient.cpp:53:0:
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:73:4: error: #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
   #error For RASPBERRY_PI_PICO_W board using CYW43439 WiFi only
    ^
In file included from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/lwip/lwip/opt.h:51:0,
                 from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/lwip/lwip/ip_addr.h:40,
                 from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/tcpip_adapter/tcpip_adapter.h:45,
                 from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/esp32/esp_event_legacy.h:23,
                 from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/esp_event/esp_event.h:26,
                 from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/esp32/esp_event_loop.h:22,
                 from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/WiFi/src/WiFiGeneric.h:27,
                 from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/WiFi/src/WiFiSTA.h:28,
                 from /home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/WiFi/src/WiFi.h:32,
                 from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:84,
                 from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/SyncClient.cpp:53:
/home/torys/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/lwip/lwipopts.h:326:0: warning: "TCP_MSS" redefined
 #define TCP_MSS                         CONFIG_TCP_MSS
 ^
In file included from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/SyncClient.h:60:0,
                 from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/SyncClient.cpp:52:
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/async_config.h:35:0: note: this is the location of the previous definition
Archiving .pio/build/mks_tinybee/lib409/libWiFi.a
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
 ^
Indexing .pio/build/mks_tinybee/lib409/libWiFi.a
Compiling .pio/build/mks_tinybee/libcbe/ESP Async WebServer/WebResponses.cpp.o
Compiling .pio/build/mks_tinybee/libcbe/ESP Async WebServer/WebServer.cpp.o
Compiling .pio/build/mks_tinybee/libcbe/ESP Async WebServer/port/SHA1Builder.cpp.o
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/async_config.h:35:25: error: 'MBED_CONF_LWIP_TCP_MSS' was not declared in this scope
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/SyncClient.h:86:34: note: in expansion of macro 'TCP_MSS'
     SyncClient(size_t txBufLen = TCP_MSS);
                                  ^
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/async_config.h:35:25: error: 'MBED_CONF_LWIP_TCP_MSS' was not declared in this scope
   #define TCP_MSS       MBED_CONF_LWIP_TCP_MSS        //(1460)
                         ^
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/SyncClient.h:87:55: note: in expansion of macro 'TCP_MSS'
     SyncClient(AsyncClient *client, size_t txBufLen = TCP_MSS);
                                                       ^
Compiling .pio/build/mks_tinybee/lib5d7/DNSServer/DNSServer.cpp.o
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:331:8: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   addr.addr = ip;
        ^
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::connect(const char*, uint16_t)':
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:390:42: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
     returnValue = connect(IPAddress(addr.addr), port);
                                          ^
In file included from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:109:0,
                 from .pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:104:
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:393:73: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
     ATCP_LOGDEBUG3("connect: dns_gethostbyname => IP =", IPAddress(addr.addr), ", returnValue = ", returnValue ? "TRUE" : "FALSE");
                                                                         ^
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W_Debug.h:87:128: note: in definition of macro 'ATCP_LOGDEBUG3'
 #define ATCP_LOGDEBUG3(x,y,z,w)  if(_ASYNCTCP_RP2040W_LOGLEVEL_>3) { ATCP_PRINT_MARK; ATCP_PRINT(x); ATCP_PRINT_SP; ATCP_PRINT(y); ATCP_PRINT_SP; ATCP_PRINT(z); ATCP_PRINT_SP; ATCP_PRINTLN(w); }
                                                                                                                                ^
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp: In member function 'bool AsyncClient::operator==(const AsyncClient&)':
.pio/libdeps/mks_tinybee/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.cpp:468:66: error: 'ip_addr_t {aka struct ip_addr}' has no member named 'addr'
   return (_pcb != NULL && other._pcb != NULL && (_pcb->remote_ip.addr == other._pcb->remote_ip.addr) &&
                                                                  ^
compilation terminated due to -fmax-errors=5.
*** [.pio/build/mks_tinybee/lib7fa/AsyncTCP_RP2040W/AsyncTCP_RP2040W.cpp.o] Error 1
Compiling .pio/build/mks_tinybee/lib678/AsyncUDP/AsyncUDP.cpp.o
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mks_tinybee/lib7fa/AsyncTCP_RP2040W/AsyncPrinter.cpp.o] Error 1
cc1plus: warning: unrecognized command line option '-Wno-register'
cc1plus: warning: unrecognized command line option '-Wno-register'
*** [.pio/build/mks_tinybee/lib7fa/AsyncTCP_RP2040W/SyncClient.cpp.o] Error 1
*** [.pio/build/mks_tinybee/lib7fa/AsyncTCP_RP2040W/AsyncTCP_RP2040W_buffer.cpp.o] Error 1
================================================================================ [FAILED] Took 4.98 seconds ================================================================================

Environment    Status    Duration
-------------  --------  ------------
mks_tinybee    FAILED    00:00:04.977
=========================================================================== 1 failed, 0 succeeded in 00:00:04.977 ===========================================================================

```

theyeetmaster992, 2025-07-05:
Get same errors on this bugfix version about GCC and AsyncTCP for tinybee

Pattis74, 2025-07-09:
Look at here: [https://github.com/makerbase-mks/MKS-TinyBee/issues/125](url)